Mshivam2409/rustduino #54: 7 Segment Display

Mshivam2409, 2021-07-19:
- [ ] Read and implement 7 segment display using digital pins. Please note that implementation must be generic i.e. please use `core::hal` instead of `core::atmega328p::hal` by using features. Make necessary changes in the code for the names to be the same and remember to discuss with the other team.
- [ ] Write an example.

dkjha20, 2021-07-19:
As our team has finalized HAL and the code is pretty same with HAL of atmega328p we ( sub team B team atmega 2560p ) would like to work on this. I think we can discuss the necessary changes in code for uniformity in HAL with "sub team B team atmega 328p" which would be working on issue #53 SN74HC164 8 bit shift registers as pointed out.
CC - @nikhilg2121 @3007-Ani-Sha @Libra-SKY

tulikashukla, 2021-07-19:
Our team will take up this issue.
Team members: Tulika, Sahil, Prateek

Mshivam2409, 2021-07-29:
Closed with #68

closed 2021-07-29 by Mshivam2409